adafruit/Adafruit_CircuitPython_PCA9685 #1: Can I use this library on the raspberry pi

opemipoVRB, 2017-07-13:
I want to control things like duty cycle and all but the library for the raspberry pi doesn't seem to have that would this work?

tannewt, 2017-07-13:
No, sorry. We'd like to build a layer underneath our CircuitPython libraries so they can run on raspberry pi but we haven't yet.

opemipoVRB, 2017-07-13:
So how can I control the speed of my servo on raspberry pi with the PCA9685 16channel 12 bit PWM LED & servo driver

opemipoVRB, 2017-07-13:
Any Suggestion ?

tannewt, 2017-07-13:
This isn't the right place to get support for it (the forums are). However, it looks like you should use `set_pwm_freq` to adjust the length of the PWM on cycle. For analog servos, that should be enough. I don't think they care about the overall frequency. They just care about the length of the on pulse.

Please follow up in the support forums if you have further questions: https://forums.adafruit.com/viewforum.php?f=50

closed 2017-07-13 by tannewt