adafruit/Adafruit_CircuitPython_PCA9685 #29: Friendly error message to catch and avoid

kevinjwalters, 2020-06-14:


razinc, 2020-10-02:
Hello.

I want to try this issue. I have a draft of the solution [here](https://github.com/razinc/Adafruit_CircuitPython_PCA9685/blob/df919e96cb32b64dacfb57fcb841b6c83dbd2bc9/adafruit_pca9685.py#L149) but I need a few clarifications:
1. How can I test my solution is working.
2. Should I print the error instead of returning it? The return value supposed to be float but in my case it will return a string. Will that cause any issue?

Thanks.

tannewt, 2020-10-02:
@razinc Do you have any hardware you can test with? Ideally we'd raise the exception that indicates an i2c error. It may be an issue in the lower levels of Blinka.

razinc, 2020-10-03:
Hello @tannewt.

Sorry. I don't have any hardware I can test it on.

Thanks.

tannewt, 2020-10-05:
@makermelissa does this sound like a Blinka issue to you? Seems like I2C should raise an exception rather than returning zero.

makermelissa, 2020-10-05:
@tannewt I think disallowing a zero value from any register at a Blinka level would likely produce unexpected results because it might be a valid value. I think adding it to this library is the correct approach.

razinc, 2020-10-06:
Hi @tannewt, @makermelissa,

Please let me know if my code needs modifications.

Thanks.

tannewt, 2020-10-06:
@makermelissa I wasn't trying to suggest erroring on zero value. I was thinking this could be a problem where Blinka should detect and i2c error and raise an exception rather than returning zero.

PR #47 (2023-01-15): Suggesting an error message solution
resolves #29 
submitted for review / comment.

My 2 cents.  it appears the only way the prescale_reg value can go south (it's read when invoked, I think) would be for I2C to fail.  According the the specsheet for PCA9685 chip, the register should be in the range of  0x03 <= value <= 0xFF.  I've only suggested for the low end, but to be airtight do we want to go the full way?   (spec sheet p 25 of 52)
closed 2023-01-15 by tekktrik