adafruit/Adafruit_CircuitPython_PCA9685 #30: maximum recursion depth exceeded error

jrconlin, 2020-07-03:
I'm trying to wire up a PCA9685 controller board to a Adafruit STM32F405 using CircuitPython 5.3.0.

The code I'm using to test things out is based on the example code you provide.

```python
import  board
import  busio

from adafruit_pca9685 import  PCA9685
from adafruit_motor import  servo

i2c_bus = busio.I2C(board.SCL, board.SDA)
pca_board = PCA9685(i2c_bus)
pca_board.frequency = 50

print(pca_board.channels[0])

servo0 = servo.Servo(pwm_out=pca_board.channels[1])

for i in range(180):
    servo0.angle = i
```

I'm getting the following:
```
Traceback (most recent call last):
  File "code.py", line 20, in <module>
  File "adafruit_motor/servo.py", line 106, in __init__
  File "adafruit_motor/servo.py", line 46, in __init__
  File "adafruit_motor/servo.py", line 50, in set_pulse_width_range
  File "adafruit_pca9685.py", line 73, in frequency
  File "adafruit_pca9685.py", line 151, in frequency
  File "adafruit_register/i2c_struct.py", line 86, in __get__
RuntimeError: maximum recursion depth exceeded
```

Hopefully, I'm just being dumb, but even after chasing the error into the libraries I'm having a hard time spotting what's triggering the recursion.

ladyada, 2020-07-03:
@hierophect ^

jrconlin, 2020-07-03:
Bit of added info while playing in the REPL:

if I create a 
```
>>> i2c_bus = busio.I2C(board.SCL, board.SDA)
>>> pca_board = PCA9685(i2c_bus)
>>> pwm = PWMChannel(pca_board, 0)
>>> p pwm.frequency
50.0288
>>> p pwm._pca.reference_clock_speed
25000000
>>> p pwm._pca.prescale_reg
122
```
which are all expected values. 

(Manually setting the `pwm.duty_cycle` steps the motor, so yay?)

calling
```
>>> servo.Servo(pwm)
```
still throws the exception at the same location.

curiouser and curiouser.

(By the way, fully aware that this is a holiday weekend in the US, and a Friday in the rest of the world. Please go enjoy being away from the computer.)

tannewt, 2020-07-06:
Hi @jrconlin, the max recursion error can be misleading because it is thrown whenever we run out of stack space. It may not be that there is any recursion. Please try 6.0.0-alpha.1, I believe we made the stack larger on the F405 since 5.3.0.

jrconlin, 2020-07-08:
Yep! That worked. Thanks!

closed 2020-07-08 by jrconlin