adafruit/Adafruit_CircuitPython_PCA9685 #52: duty_cycle getter doesn't return what setter was given

DrRob, 2023-04-27:
Hi, why does the setter have `value + 1` in line 91?:

`value = (value + 1) >> 4`

There isn't a corresponding adjustment in the getter, so for example if I set duty_cycle to 0x7fff and then read it back I get 0x8000.

Thanks!
Rob.

tekktrik, 2023-04-28:
@tannewt Looks like this was in the original commit for the library, curious if you know or remember why the getter and setter would be different?

tannewt, 2023-04-28:
I don't remember. Sorry!

It would make more sense to read the same value back.

tekktrik, 2023-05-01:
Thanks @tannewt!

Looks like it might be a bug @DrRob, are you interested in submitting a PR to fix it?

DrRob, 2023-05-01:
I'll try, but I'm puzzled about why the `value + 1` part doesn't result in a requested duty cycle of zero actually being on for 1/4096th of the period.  (It doesn't - I checked with an oscilloscope.)

DrRob, 2023-05-01:
Actually I got that wrong - if a duty cycle of zero is requested, then `(0 + 1) >> 4` == `1 >> 4` == 0, so both the LEDn_ON and LEDn_OFF registers are set to zero, which probably explains why the output is off all the time as expected...

...HOWEVER, the datasheet says, "The LEDn_ON and LEDn_OFF count registers should never be programmed with the same values."  LEDn_OFF should be set to 0x1000 for fully off.

I'll try to fix it according to the datasheet instructions.

DrRob, 2023-05-02:
I'm a bit rusty with git but I think I've just submitted a PR for this.

closed 2023-05-22 by dhalbert