adafruit/Adafruit_CircuitPython_PCA9685 #61: Example not running on Raspberry Pi 5

pReya, 2024-03-13:
When I run any of the examples on a Raspberry Pi 5, I get this:

```
ImportError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/moritz/pwm2/test.py", line 8, in <module>
    import board
  File "/home/moritz/pwm2/.venv/lib/python3.11/site-packages/board.py", line 50, in <module>
    from adafruit_blinka.board.raspberrypi.raspi_5b import *
  File "/home/moritz/pwm2/.venv/lib/python3.11/site-packages/adafruit_blinka/board/raspberrypi/raspi_5b.py", line 6, in <module>
    from adafruit_blinka.microcontroller.bcm2712 import pin
  File "/home/moritz/pwm2/.venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm2712/pin.py", line 5, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/moritz/pwm2/.venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py
```

Any ideas?

Sumlayen, 2024-04-04:
I know it's not an optimal solution but I had to do it all with sudo.  First I had to do "sudo pip install build" so the following would work.  Follow the instructions from https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py but do all the commands as sudo.  Then when I ran my python scripts that used servos I had to use sudo.  I hate this workaround but it does work.  I could not get it to work in a conda virtual environment.  Hopefully it's addressed by smarter folks than me in the future.

lukelmg, 2024-05-14:
> I know it's not an optimal solution but I had to do it all with sudo. First I had to do "sudo pip install build --break-system-packages" so the following would work. Follow the instructions from https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py but do all the commands as sudo with the "--break-system-packages" flag for any pip installs. Then when I ran my python scripts that used servos I had to use sudo. I hate this workaround but it does work. I could not get it to work in a conda virtual environment. Hopefully it's addressed by smarter folks than me in the future.

These steps worked for me however yes, I would like the ability to run in a conda virtual environment.